pcdshub/ioc-common-ads-ioc #21: Autosave doesn't work

ZLLentz, 2020-02-13:
https://github.com/pcdshub/ads-ioc/blob/c4720149a1f0af1a995646a621324e5cc7a81176/iocBoot/templates/st.cmd.template#L115 changes to the wrong top

https://github.com/pcdshub/ads-ioc/blob/c4720149a1f0af1a995646a621324e5cc7a81176/iocBoot/templates/st.cmd.template#L133 references a file that is never built

klauer, 2020-02-14:
(1) We're calling "IOC_TOP" where the `st.cmd` is - probably a bad name. Or do you mean it didn't even `cd` there?
(2) `makeAutosaveFiles` needs to be called

ZLLentz, 2020-02-14:
Expanding on (1): here we need to cd into the directory that has:
```
dbLoadRecords("db/iocSoft.db", "IOC={{prefix}}")
dbLoadRecords("db/save_restoreStatus.db", "P={{prefix}}{{delim}}")
```
Which is the ads module top

(2) is missing from the default `make` I think

ZLLentz, 2020-02-18:
closed by #22

closed 2020-02-18 by ZLLentz