pcdshub/ioc-common-ads-ioc #26: Motor SPDB left at default value of 1

ZLLentz, 2020-02-19:
This has caused some confusion. The SPDB field is the setpoint deadband. The IOC will ignore moves that are less than this value. There is no obvious way to configure this as-is because it's missing from https://github.com/pcdshub/m-epics-ethercatmc/blob/master/EthercatMCApp/Db/EthercatMC.template. I think we should forcibly set this to zero before the pass0 restore...

ZLLentz, 2020-02-19:
Actually, it may be better to make a modification to that template file to also configure autosave/archiving...

klauer, 2020-02-26:
Looks like the SPDB field is set to the precision using this record:
```
record(ao, "$(PREFIX)$(MOTOR_NAME)-CfgSPDB_")
{
    field(PREC, "$(PREC)")
    field(DOL,  "$(PREFIX)$(MOTOR_NAME)-CfgSPDB CP")
    field(OUT,  "$(PREFIX)$(MOTOR_NAME).SPDB")
    field(OMSL, "closed_loop")
}
```

ZLLentz, 2020-02-27:
Does the `$(PREFIX)$(MOTOR_NAME)-CfgSPDB` record (without the trailing `_`) exist at all though?