pcdshub/ioc-common-ads-ioc #45: Repo + tags out of sync with NFS

klauer, 2020-05-20:
@mcb64 if possible, would you mind using pull requests in the future for updates?

We primarily follow the pull request/review model for the rest of our projects.

klauer, 2020-05-20:
`/afs/slac/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git` has:

```
commit 5253effb9265dd37b22ebf3ce9f03a665b660a34
Author: Michael Browne <mcbrowne@slac.stanford.edu>
Date:   Tue Apr 21 09:24:17 2020 -0700

    Damn it, update version number!

commit b2ededafa61e04ac7476e335082b50e638247bd6
Author: Michael Browne <mcbrowne@slac.stanford.edu>
Date:   Mon Apr 20 16:44:36 2020 -0700

    Move to latest releases.

commit cefd4ac530bddfd5bac414ee0f85142efb8c833d
Author: Michael Browne <mcbrowne@slac.stanford.edu>
Date:   Fri Apr 3 13:04:50 2020 -0700

    In-progress checkin.

commit b777b3f4194f36b987755f0cc87a98e9a4ebe9be
Merge: a79339b 45cee94
Author: K Lauer <klauer@users.noreply.github.com>
Date:   Tue Mar 3 14:10:49 2020 -0800

    Merge pull request #35 from klauer/add_pytmc_version_check

    ENH/FIX: check pytmc version prior to building
```

making  b777b3f4194f36b987755f0cc87a98e9a4ebe9be the point at which the repositories diverage

mcb64, 2020-05-20:
Not at all, sorry, I'm a git noob... is there a page about the workflow somewhere?
    --Mike

klauer, 2020-05-20:
@mcb64 it's OK, as it's partly my fault for not keeping our local mirror up-to-date. Docs on the branch/PR https://pcdshub.github.io/development.html - but the basic gist is "make a feature branch, open a pull request into master, review + merge, tag".

I _think_ what we might want to do here is leave R0.2.3 as-is, but then bring in your changes to the our primary repository (here on github) as a separate PR. I'm open to suggestions though (@ZLLentz?)

mcb64, 2020-05-20:
I'd be happy to do the merge / pull request thing... just let me know what version / repository / branch I should be merging into.

And use *really* tiny words... ;-)
    --Mike

closed 2020-05-21 by klauer

klauer, 2020-05-21:
We're back in sync: `/afs/slac/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git`
is at cd19ecee43ab948878f6c819ca95ae9370eee0e9 (merge of #46) 👍

PR #46 (2020-05-21): Merge in diverging afs/GIT_TOP master
Closes #45 

TODO, I think:
* Keep R0.2.3 as-is on AFS
* Base next release off of pcdshub/ads-ioc
* Force push to GIT_TOP master

Open to alternatives as well