pcdshub/ioc-common-ads-ioc #47: Clarify primary repositories

klauer, 2020-05-21:
There's (understandable) confusion with all of our repositories, which can of course lead to synchronization issues.

On pcdshub (public):
* https://github.com/pcdshub/ads-ioc
* https://github.com/pcdshub/twincat-ads
* https://github.com/pcdshub/m-epics-ethercatmc (ESS naming)

On slac-epics (public):
* https://github.com/slac-epics/twincat-ads
* https://github.com/slac-epics/ethercatmc (SLAC naming)

On afs $GIT_TOP (SLAC private):
* `/afs/slac/g/cd/swe/git/repos/package/epics/modules/twincat-ads.git `
* `/afs/slac/g/cd/swe/git/repos/package/epics/modules/ethercatmc.git`
* `/afs/slac/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git`

My opinion is that pcdshub should remain our primary development platform (point of collaboration, where issues and pull requests go), slac-epics should remain our module repository mirror (everything in one place). 

Though it may differ from normal workflows, the above also means that GIT_TOP becomes a mirror of pcdshub for these respective repositories.

What can we do from here - open-ended ideas/thoughts:
* For slac-epics, perhaps we can recreate the repositories and make them forks of pcdshub?
* For GIT_TOP, I'm not sure we can do anything to keep people from pushing to it directly
* Keep the repository names consistent (ethercatmc / m-epics-ethercatmc?)

hhslepicka, 2020-05-21:
For the GIT_TOP Zhou could deploy a server-side pre-receive book and deny the push with a message to notify the user of he proper workflow.

klauer, 2020-05-21:
Interesting - have details on how to do that @hhslepicka?

hhslepicka, 2020-05-21:
@klauer yes! I will get back to you on internal Slack chat.

klauer, 2020-05-21:
OK, this works pretty well, thanks @hhslepicka!

```
[klauer@psbuild-rhel7-01  ads-ioc]$ git push gittop hook_test
Total 0 (delta 0), reused 0 (delta 0)
remote:
remote:
remote: Please open a pull request here: https://github.com/pcdshub/ads-ioc
remote: If this repository is out of sync with GitHub, do the following:
remote:  * cd /afs/slac.stanford.edu/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git && git pull origin master
remote:
remote:
To /afs/slac/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git/
 ! [remote rejected] hook_test -> hook_test (pre-receive hook declined)
error: failed to push some refs to '/afs/slac/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git/'
```

In `/afs/slac.stanford.edu/g/cd/swe/git/repos/package/epics/ioc/common/ads-ioc.git/hooks/pre-receive`:

```
#!/bin/bash                                                                          
echo ""                                                                              
echo ""                                                                              
echo "Please open a pull request here: https://github.com/pcdshub/ads-ioc"           
echo "If this repository is out of sync with GitHub, do the following:"              
echo " * cd $PWD && git pull origin master"                                          
echo ""                                                                              
echo ""                                                                              
exit 1                                                                               
                                                                                     
is_push_to_github                                                                    
exit $?                                                                              
```

hhslepicka, 2020-05-21:
Just a note that is very important... Usually when at SLAC someone uses the scripts to clone a github repo at our internal AFS hosted GIT it will not come with this hook... so whoever clones it internally must remember to add that... 😞

klauer, 2020-05-21:
Good point, @hhslepicka 👍 

OK, I think with these changes we can consider this closed:
* slac-epics now shows that it has forked from pcdshub ethercatmc, twincat-ads, and no longer has an 'issues' section
* pre-receive hooks for the repositories in question are in place on AFS
* renamed pcdshub/m-epics-ethercatmc to fall in line with the naming in slac-epics and afs/git_top

closed 2020-05-21 by klauer