pcdshub/ioc-common-ads-ioc #54: Logfile inconsistencies in dev mode

ZLLentz, 2020-06-18:
In dev mode, the ioc still tries to write the logfiles to the /reg/d/iocData directory instead of the user's home area, and with a different ioc name. Compare the following directories:
`~jjoshi/iocData/ioc-plc-kfe-rix-vac`
`/reg/d/iocData/ioc-kfe-rix-vac`

klauer, 2020-06-18:
I'm guessing that means we don't have logs for debugging this instance? ☹️

ZLLentz, 2020-06-18:
We do, they are just in the oddly wrong-named iocData location :shrug:

klauer, 2020-06-23:
The IOC vs IOCNAME mix-up caused the incorrect path "stem" (hopefully closed by #58 )

The `/reg/g/iocData` location must be baked-in somewhere else, as the logging configuration happens by way of the parent process. Not sure what we can do there.

ZLLentz, 2020-06-23:
It looks like ioc manager sets `IOC_DATA` to `/reg/g/iocData` and is also responsible for managing the logfile.