pcdshub/ioc-common-ads-ioc #56: default sample time is too slow?

rajanplumley, 2020-06-23:
[image]

ZLLentz, 2020-06-23:
Thanks for posting this Raj, it's one of those things that is "known" vaguely but never got properly documented. This is a consequence of us having crazy numbers of PVs in our IOCs. I suspect we want to be smarter about the sample time we choose rather than always bumping to 1Hz.

rajanplumley, 2020-06-23:
Ok cool.  Is there any way to know or test if the sample time is too small, so it can step it up until it reaches a minimally satisfactory sample time?

ZLLentz, 2020-06-23:
The IOC can tell if the current sampling time is "too fast" by seeing if it has enough time to process all the PVs before the next scheduled update. I do not know if it is possible to continually change the parameter and optimize it, or if it needs to be set once at the start of the program as the module is currently written.

The consequence of not increasing the sampling time from 50ms is that you'll get your PVs back at arbitrary intervals, e.g. it will go "as fast as it can" and finish whenever it does. Having the sampling time too low can also theoretically impact PLC performance, but I'm not sure if this is a leading-order concern compared to the IOC process getting overloaded.