pcdshub/ioc-common-ads-ioc #68: Repeated errors: "callbackRequest: cbLow ring buffer full"

klauer, 2020-11-19:
Seen in the latest version of https://github.com/pcdshub/lcls-plc-tmo-vac/tree/master/iocBoot/ioc-PLC-TMO-VAC
(not yet pushed as of writing: `/reg/g/pcds/epics-dev/jyin/gitProjects/lcls-plc-tmo-vac`)

Seems like we may need to increase the callback queue size by way of `callbackSetQueueSize` (default may be 2000?)

ZLLentz, 2020-11-20:
I wonder if this a race condition based on whatever else is running on the server...

Should we set this to something like the number of PVs if we have more than 2000 PVs?

According to EPICS tech talk the main drawback is more RAM used (shouldn't be a big deal):
https://epics.anl.gov/tech-talk/2017/msg00811.php

klauer, 2020-11-20:
Race does seem a definite possibility, based on what Jing had seen when switching to an IOC host with less load.

Agreed on setting it to something like the number of PVs (or proportional to it). We had another parameter we wanted to do proportionally for asyn, didn't we... Hmm... Will take care of these tomorrow.

PR #69 (2020-11-20): ENH: customize callback queue size
## Unimportant details

Going through this, I see why we didn't dynamically set the ADS parameters. The st.cmd generation and database generation are broken up into a few steps which don't really know about the other, making it more confusing than it needs to be. That can be redone down the line without incurring much risk except keeping backward-compatibility without annoying everyone too much.

For some odd reason the `dbloads.py` is Python 2-friendly. I'm keeping it that way for now.

## Important details
* Provide enough `Makefile`-level hooks to configur [... cut]
closed 2020-11-20 by klauer