pcdshub/ioc-common-ads-ioc #70: Determine task count dynamically for taskinfo.db

klauer, 2020-12-02:
Some users are seeing these messages at startup:
```
2020/12/01 17:04:41.873 adsAsynPortDriver:adsGetSymInfoByName: Get symbolic information failed for TwinCAT_SystemInfoVar                               List._TaskInfo[2].TaskName with: ADSERR_DEVICE_INVALIDARRAYIDX (0x721)
devAsynInt8Array::initCommon, PLC:las_bts:TaskInfo:2:TaskName drvUserCreate failed
```

It appears we don't always have 2+ tasks?

ref https://github.com/pcdshub/ads-ioc/blob/master/app/Db/TwinCAT_TaskInfo.db

ZLLentz, 2020-12-03:
Do we read it from the tmc/project? Or is there some ADS interface for getting the number of tasks?

klauer, 2020-12-03:
I took a brief look into the TMC file and didn't see anything regarding tasks; but I assume it's because it's a fairly dynamic thing?

ZLLentz, 2020-12-04:
I think the number of tasks is predefined in the project, so the information should be in there somewhere

klauer, 2020-12-04:
Hmm, you're right - this is also useful information for the summary pages:

```
In [9]: list(project.find(Tasks))
Out[9]: [<Tasks children=[<Task name='PlcTask' attributes={'Id': '5', 'Priority': '20', 'CycleTime': '100000', 'AmsPort': '350', 'AdtTasks': 'true'} children=[<Name text='PlcTask'>]>]>]
```

However, for `lcls-plc-kfe-motion` I'd have expected to find more than just one (right?)

ZLLentz, 2020-12-04:
Probably only one PLC task, but there should be a few NC tasks

ZLLentz, 2020-12-04:
I suspect most of our PLCs have either 1 task or 2 (fast task and slow task)

klauer, 2022-07-08:
I think I should tackle this as part of #86 - need to look into this again

PR #86 (2022-09-06): REF: use pytmc template to generate everything; fix lots of things
## Changes

* pytmc `>=2.14.0` now required
* Use `pytmc template` from upcoming pytmc release to generate:
   * st.cmd
   * Database files
   * Archiver files
* Removes `load_plc_databases.cmd` - everything is in `st.cmd` now
* Adds a significant amount of version information to the header
* Removes motion axes from IOC without pragma (*)
* Style-wise, makes macro usage consistent in st.cmd: `$()`
* Adds ability to easily specify the PLC hostname with an IOC Makefile override
* `make build` should no longer copy to iocData dir [... cut]
closed 2022-09-06 by klauer