pcdshub/ioc-common-ads-ioc #85: Add alternative method for specifying PLC IP address

klauer, 2022-07-06:
## Goal

* Allow a PLC to be relocated to a different hutch with a change of IP address via netconfig
* Do not require the AMS Net ID of the PLC to be reconfigured for this move

## st.cmd location

https://github.com/pcdshub/ads-ioc/blob/b2cb4a015cb062dc2be62a38f6418c7551e92f76/iocBoot/templates/st.cmd.template#L21

## pytmc source

We still don't have a way to get the PLC hostname directly from the project. All that's stored is the AMS Net ID in the project, from which we guess the IP:

https://github.com/pcdshub/pytmc/blob/233acfb6da27162d34db8de7658684d203a4caad/pytmc/parser.py#L510-L518

## Options

### An override?

Add an override to `pytmc stcmd` for the IP address:
```make
# IOC Makefile
PLC_IP_ADDRESS:=1.2.3.4

# ads-ioc Makefile.base (pseudo make syntax because it's complicated)
st.cmd: _check_versions
	@echo "Executing pytmc stcmd: creating st.cmd..."
	cd "$(BUILD_PATH)" && \
		pytmc stcmd \
		--name "$(IOC_NAME)" \
		--plc "$(PLC)" \
		--template-path "$(TEMPLATE_PATH)" \
		--template "$(STCMD_TEMPLATE)" \
		--hashbang "$(BINARY_PATH)" \
		--force-ip "$(PLC_IP_ADDRESS)" \
		--only-motor \
		-p "$(PREFIX)" \
		$(PYTMC_OPTS) \
		"$(call pyabspath,$(PROJECT_PATH))" > st.cmd
```

May be worth jumping to `pytmc template` as it's been planned for a while https://github.com/pcdshub/pytmc/issues/209

### A second sed step?

```make
# IOC Makefile
PLC_IP_ADDRESS:=1.2.3.4

# ads-ioc Makefile.base (pseudo make syntax)
build:
    ... normal build step
    if PLC_IP_ADDRESS:
        sed -i -e "s/{{ pytmc_ip }}/$PLC_IP_ADDRESS/g" st.cmd
```

ZLLentz, 2022-07-06:
I think the end goal should be allowing the user to specify `plc_ip=my-plc-hostname` in the `Makefile` if they want to- the IOC should be able to run using this string, rather than an explicit IP, right? Then your IOC build becomes resistant to adjustments to the IP configurations so long as your ams net id remains static. Options to `pytmc stcmd` seem like an appropriate path to fulfilling this.

klauer, 2022-07-06:
> if they want to- the IOC should be able to run using this string, rather than an explicit IP, right?

Yes, asyn is backing ads-ioc and it does resolve hostnames without issue. 
The above wording and example was very poor, but working with hostnames rather than IPs is definitely intended.

PR #86 (2022-09-06): REF: use pytmc template to generate everything; fix lots of things
## Changes

* pytmc `>=2.14.0` now required
* Use `pytmc template` from upcoming pytmc release to generate:
   * st.cmd
   * Database files
   * Archiver files
* Removes `load_plc_databases.cmd` - everything is in `st.cmd` now
* Adds a significant amount of version information to the header
* Removes motion axes from IOC without pragma (*)
* Style-wise, makes macro usage consistent in st.cmd: `$()`
* Adds ability to easily specify the PLC hostname with an IOC Makefile override
* `make build` should no longer copy to iocData dir [... cut]