pcdshub/ioc-common-ads-ioc #89: autosave request file directory is probably in a bad location

klauer, 2022-08-08:
https://github.com/pcdshub/ads-ioc/blob/b2cb4a015cb062dc2be62a38f6418c7551e92f76/iocBoot/templates/st.cmd.template#L133

I saw for `ioc-kfe-vac` that `/reg/g/pcds/epics-dev/ioc/kfe/lcls-plc-kfe-vac/ioc/ioc-kfe-vac/autosave` was not writable by the IOC, which it needs to be for `makeAutosaveFiles`. This means I needed to make it writable for `ps-ioc`, which I think is probably undesirable.

Options, in order of preference:
1. Make the request directory $IOC_DATA as well
2. Adjust the `Makefile` to `chown $USER:ps-ioc autosave`. This would require the IOC builder to be part of the `ps-ioc` group
...
...
100. Adjust the `Makefile` to set very relaxed permissions on `autosave`

ZLLentz, 2022-08-10:
For the other IOCs we stash these in `$IOC_DATA/$IOC_NAME/autosave`, I think

klauer, 2022-08-10:
I seem to recall that storing `.req` files in `/cds/data` wasn't the standard and I think I only now understand why, looking back:
1. For our non-ads-ioc IOCs, most `.req` files were built with the templating mechanism. This means they are build artifacts and were commonly stored in `iocBoot/ioc/autosave`
2. ads-ioc employs `info`-tag based `.req` file generation - which is easier and just generally better (in my opinion). These are then not build artifacts but runtime-generated artifacts. Regardless, we stored them as in (1).
3. Almost no other IOCs (except for external ones or areaDetector-based ones) use `makeAutosaveFiles` (in my 1 minute search)

So because of the misunderstanding, we stored them in the wrong spot. I feel more confident to switch this over. Perhaps even in the massive PR #86 for this afternoon.

ZLLentz, 2022-08-10:
Right, that makes sense- we stored them in the same location as the other IOCs based on file type/content rather than storing them based on creation circumstances (at `make` vs at runtime). But since we're creating them at runtime they should go in the runtime output directory.

PR #86 (2022-09-06): REF: use pytmc template to generate everything; fix lots of things
## Changes

* pytmc `>=2.14.0` now required
* Use `pytmc template` from upcoming pytmc release to generate:
   * st.cmd
   * Database files
   * Archiver files
* Removes `load_plc_databases.cmd` - everything is in `st.cmd` now
* Adds a significant amount of version information to the header
* Removes motion axes from IOC without pragma (*)
* Style-wise, makes macro usage consistent in st.cmd: `$()`
* Adds ability to easily specify the PLC hostname with an IOC Makefile override
* `make build` should no longer copy to iocData dir [... cut]
closed 2022-09-06 by klauer