pcdshub/ioc-common-ads-ioc #90: IOC name inconsistencies with IOC manager

klauer, 2022-08-11:
There have been several cases to my knowledge of mismatched `$(IOC)` name compared to what IOC Manager has assigned as its ID.

This is a big problem since:
* IOC manager will write log files and things to one directory
* And ads-ioc will assume things are in another directory

This leads to confusion and potentially incorrect behavior in the end.

This is set here: https://github.com/pcdshub/IocManager/blob/fabc34c31e4fbb14c9d3b746474f42b3ef9e3688/startProc#L24
As part of the [full process](https://github.com/pcdshub/ioc_machine_core/blob/master/.docs/script_order.png?raw=true): https://confluence.slac.stanford.edu/display/PCDS/Detailed+Soft+IOC+boot+process#DetailedSoftIOCbootprocess-startProc

So we should be able to rely on $(IOC) being set for us externally. We could instead do:
`epicsEnvSet("IOC", "$(IOC=iocname)")`
to set a default in case it's being run _without_ iocmanager (which can be useful in development/testing situations).

<details>

```
epics> epicsEnvShow GITHUB_USER
GITHUB_USER=klauer
epics> epicsEnvSet("GITHUB_USER", "$(GITHUB_USER=asdf)")
epics> epicsEnvShow GITHUB_USER
GITHUB_USER=klauer
```
and
```
epics> epicsEnvShow GITHUB_USER_A
GITHUB_USER_A is not an environment variable.
epics> epicsEnvSet("GITHUB_USER_A", "$(GITHUB_USER_A=asdf)")
epics> epicsEnvShow GITHUB_USER_A
GITHUB_USER_A=asdf

```
</details>

ZLLentz, 2022-08-11:
Oh this is annoying

Letting the ioc manager set the ioc name seems reasonable. I'm trying to figure out why this isn't an issue for other IOCs but I'm coming up blank. Usually this gets set to the directory name under `iocBoot` in ioc manager since this is how the ioc manager locates the st.cmd file. So we can possibly check the mismatch by comparing our iocname to the directory name? But this assumes things are being done the usual way (ioc path is `/some/long/path/v1.0.0` and ioc name is set to the subdirectory name)

klauer, 2022-08-11:
I think there's no way we can do the check, unless we shell out to an external script. There's no real control flow options in `st.cmd` after all

I put this in the PR: https://github.com/pcdshub/ads-ioc/pull/86/commits/184e609c5c4c02310fbd22071ad13d145256ffa8
I think ideally we'd want a verification that they match, but in the meantime deferring to IOC Manager to set it if possible seems good enough to me?

ZLLentz, 2022-08-11:
Yes, I think giving the IOC manager control makes the most sense.

PR #86 (2022-09-06): REF: use pytmc template to generate everything; fix lots of things
## Changes

* pytmc `>=2.14.0` now required
* Use `pytmc template` from upcoming pytmc release to generate:
   * st.cmd
   * Database files
   * Archiver files
* Removes `load_plc_databases.cmd` - everything is in `st.cmd` now
* Adds a significant amount of version information to the header
* Removes motion axes from IOC without pragma (*)
* Style-wise, makes macro usage consistent in st.cmd: `$()`
* Adds ability to easily specify the PLC hostname with an IOC Makefile override
* `make build` should no longer copy to iocData dir [... cut]
closed 2022-09-06 by klauer